Eryone/STM32 #3: PID autotune freezes printer

sihawken, 2021-03-24:
I cannot run a PID autotune procedure with this printer. Both the LCD screen halts, and the serial communication stops.

Eryone, 2021-04-15:
It should be the printer motherboard problem.
On our test, it works.

Another user replaced it and it fixed.

stormz85, 2021-04-15:
Hi !

when i do that my printer get freezed i have to reboot to use it
mother board was changed and the issue is still there.
Updated also with the marlin STM32 on the Eryone Github same issue.

Here is the command i use : M303 E0 S200 C8

Dawood008, 2021-04-26:
Hi Storm, try to check firmware from this link = https://www.eryone.com/forum/viewtopic.php?f=6&t=439
Try PID after uploading the firmware, share details with us

Eryone, 2021-05-24:
Hi, the PID freezes problem fixed at the new version of the firmware
2.0.8
You can go to download the new firmware and upload it.

closed 2021-05-24 by Eryone