Eryone/STM32 #5: Marlin command M117 seems to start Auto Bed Leveling on ER20

MacPro-de, 2021-04-06:
Hello, we used the firmware 2.0.5.3 from this Github page and tried the following with the prepared firmware as well as with a version, that we compiled by ourselves.
The printer is connected to a Raspberry Pi with Octoprint.
With both versions it happens, that, when we issue a M117 command via the terminal in Octoprint, then in 90% of the cases the printer starts the Autobed Leveling process (as we would have issued a G29 command). In 5% of the cases nothing happens. Only in the other 5% the M117 command really displays our text on the LCD.
Is this a known issue? Any help would be much appreciated.

ARM-Mac, 2021-04-08:
i have the same Problem.

Eryone, 2021-04-15:
We have update the bug and now it fixed!
Thx!

closed 2021-04-15 by Eryone