Eryone/STM32 #8: Klipper firmware for ER20

raumin, 2021-08-02:
Hi,

Do you plan to add instructions on how to compile klipper firmware and flashes it onto the ER20?

i tried to make a firmware by following klipper instructions and the config here for ER20 https://github.com/KevinOConnor/klipper/tree/master/config but i failed, generate firmware seems to be ok but flash seems to be more difficult...

"To use this config, the firmware should be compiled for the
# STM32F103 with a 16KiB HID Bootloader."

It doesnt work for me, i tried 28KiB, i tried USART1/2/3 and nothing.
I tried to upload .bin file directly from windows with your flash tool but no success.

Could be great ;)
Greetings.

Christdecenon, 2022-11-28:
Hello everyone, like Raumin, I'm stuck in the same place. If someone passes by here and knows the solution ^^
Thanks a lot ;-)

raumin, 2022-12-04:
> Hello everyone, like Raumin, I'm stuck in the same place. If someone passes by here and knows the solution ^^ Thanks a lot ;-)

Hi, i succeeded with Mainsail, worked like a charm. 
There is just a random problem with our ER20, some people like me has buggy screen with Klipper and some people doesnt have the buggy screen... Anyway, it's not a problem, just the screen is unreadable.

If you need help, there is a messenger group available to help.

Greetings

NightShifterOne, 2023-05-05:
> Anyway, it's not a problem, just the screen is unreadable.
> If you need help, there is a messenger group available to help.
> 
 Hello raumin,
 
 i have the same problem with my ER20 Display.
 It looks like it has the wrong resolution?
 Is there a solution for this issue? 

Thanx and regards...

raumin, 2023-05-05:
hey @NightShifterOne ,

LCD glitches fixed for me with this : https://github.com/Klipper3d/klipper/issues/5089
Replace this two files https://github.com/Stout69/For-Klipper ==> /src/lcd_st7920.c and /klippy/extras/display/st7920.py

# make clean
# make

flash with the .bin and works perfectly.
Hope this helps with you.

But with this, your klipper will be notified as "DIRTY", not a problem but you will need to replace this 2 files if you update klipper.

From Demetrius on the messenger group when i asked howto avoid "dirty" notification :

> You probably can get around this by configuring you klipper repo to point to this guys repo instead if he keeps it up to date.
> https://github.com/Stout69/For-Klipper
> Else, you would need to create your own fork of Klipper, integrate his changes and then configure the Klipper folder on the Pi to look at your repo

Regards